stineje/CharLib #2: LICENSE might be wrong?

mithro, 2022-11-22:
In the README you say;
> This open-source characterization tool is modified from the libretto characterization tool found here : https://github.com/snishizawa/libretto.

libretto is licensed under the GPL v2.0 license (see https://github.com/snishizawa/libretto/blob/main/LICENSE). If this code is a derivative work of libretto, then it needs to also be released under the same license.

As well, if this tool is derivative work of libretto, then we should be contributing the changes to libretto rather than creating a fork.

stineje, 2022-11-22:
done.

mithro, 2022-11-22:
If this is really suppose to be a derivative work of libretto (and from what I can see it is), then I can rework your repository history so it preserves this information / relationship.

infinitymdm, 2023-05-25:
This issue appears to be fixed.

@stineje is there any reason for @mithro not to rework our repo history to preserve the linkage to libretto? We mention it right there on the first line of our documentation anyways.

closed 2023-05-25 by stineje

proppy, 2023-05-26:
looks like the history is indeed related to https://github.com/snishizawa/libretto, see original commit here: https://github.com/stineje/CharLib/commit/41b5e87e8f721dc82e3e4c6343635a345aeaae26

Even if the project has diverged quite a lot since then, it'd be nice to rebase the history on top of the original repo at the point of time it was forked to show the code relationship between the two project.

Additionally the project README currently only reference the paper, it'd be nice to also point to the GitHub repository.

infinitymdm, 2023-05-27:
@stineje and I discussed this, and we don't want to link the git history since we do not intend to sync commits from upstream.

As for whether to link to libretto: I think it would be nice to do so, but I'll leave the final decision on that to Dr. Stine.