stineje/CharLib #11: Add option to show plots for characterization

infinitymdm, 2023-05-26:
Add options to generate plots that show the details of what's happening during characterization, as well as a nicer view of characterization results.

@stineje and I discussed the following plots as a good starting point:

- For timing characterization
    - [ ] I/O voltages vs. time
    - [ ] Transport and Propagation delay vs fanout (displayed as a 3D surface plot)
- For power characterization
    - [ ] Power vs. fanout

infinitymdm, 2023-05-30:
Since python SPICE interfaces exist with great support for plotting libraries like matplotlib, we're also going to take this opportunity to switch the backend over to PySpice.

The current method generates spice files in a mostly-manual process, then writes commands to a shell script. Switching to PySpice should make our code a lot more readable and intuitive, plus we'll have quite a bit more insight into what's happening in our spice simulations.

thesourcerer8, 2023-08-14:
You can use libertyviz from librecell for 3D visualisation of the plots: https://codeberg.org/librecell/lctime

closed 2023-10-03 by infinitymdm