stineje/CharLib #18: Goals and Intended Features

infinitymdm, 2023-10-27:
This issue tracks major goals and features for CharLib. Other issues and PRs will be linked here as features are completed or worked towards.

- [x] Support combinational and sequential cells
- [ ] Take advantage of multithreading
- [ ] Characterize timing, power, and noise
- [ ] Provide CCSM, ECSM, and NLDM models
- [ ] Account for setup and hold time interdependence in sequential characterization
- [ ] Support HSPICE as a circuit simulator

Feedback and suggestions are, of course, welcome.

TiMo3654, 2024-08-21:
Hi @infinitymdm, this is a great project so far! Were you able to make any further progress on the points listed above? Especially the different kind of models are of great interest to me. Keep up the good work!

infinitymdm, 2024-08-21:
> Were you able to make any further progress on the points listed above? Especially the different kind of models are of great interest to me.

Unfortunately the literature on how ECSM and CCSM work seems to be pretty sparse. There may be enough info out there to figure it out, but what I've read so far only seems to hint at methods rather than actually explaining how it works. So unfortunately there's been no major progress there. 

Currently the priority is to get power characterization working for NLDM. Other models are a much bigger task that will follow later.

TiMo3654, 2025-07-29:
For the support of the spectre simulator the [pyspectre](https://github.com/AugustUnderground/pyspectre) package could be a solid choice.

However, I guess that the testbench netlist creation process in CharLib might not be so easily convertable.

infinitymdm, 2025-07-29:
> For the support of the spectre simulator the [pyspectre](https://github.com/AugustUnderground/pyspectre) package could be a solid choice.
> 
> However, I guess that the testbench netlist creation process in CharLib might not be so easily convertable.

I wasn't aware of this package. Thanks for pointing it out - may be useful in the future. 

PySpice does have the faculty to write out spice files, so it might not be too terrible to integrate this. Either way, good to know about.