stineje/CharLib #23: function parsing problem

thesourcerer8, 2023-12-11:
It seems to me that the negation operator ! which should have the highest priority isn't working properly, only when I add round brackets around the term that is negated, the error goes away:
[libresilicon-charlib.yml.txt](https://github.com/stineje/CharLib/files/13638358/libresilicon-charlib.yml.txt)

Searching for YAML files in .
Reading configuration found in "libresilicon-charlib.yml"
Prefixed unit not found for 1e-15 J and power -15
Prefixed unit not found for 1e-15 J and power -15
Traceback (most recent call last):
  File "/home/philipp/.local/bin/charlib", line 8, in <module>
    sys.exit(main())
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/run.py", line 60, in main
    args.func(args)
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/run.py", line 122, in run_charlib
    characterizer.add_cell(name, inputs, outputs, functions, **properties)
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/Characterizer.py", line 21, in add_cell
    CombinationalTestManager(name, in_ports, out_ports, functions, **kwargs)
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/TestManager.py", line 61, in __init__
    if reg_func == function:
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/functions/functions.py", line 68, in __eq__
    result = self.truth_table() == other.truth_table()
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/functions/functions.py", line 60, in truth_table
    result = self.eval(**dict(zip(self.operands, input_vector)))
  File "/home/philipp/.local/lib/python3.10/site-packages/charlib/characterizer/functions/functions.py", line 51, in eval
    f = eval(f'lambda {",".join(operands)}: int({self.expression.replace("~", " not ")})')
  File "<string>", line 1
    lambda B,A,A1,C1,C,B1: int( not (C1&( not C)&B1&B&( not A1)&( not A)|C1&C&B1&B& not A1& not A| not C1&C&B1&B& not A1& not A| not C1&C&B1&B&A1& not A|C1&C&B1&B&A1& not A|C1& not C&B1&B&A1& not A|C1& not C& not B1& not B&A1&A|C1&C& not B1& not B&A1&A| not C1&C& not B1& not B&A1&A| not C1&C&B1& not B&A1&A|C1&C&B1& not B&A1&A|C1& not C&B1& not B&A1&A|C1& not C&B1&B&A1&A|C1&C&B1&B&A1&A| not C1&C&B1&B&A1&A| not C1&C& not B1&B&A1&A|C1&C& not B1&B&A1&A|C1& not C& not B1&B&A1&A| not C1&C&B1&B& not A1&A|C1&C&B1&B& not A1&A|C1& not C&B1&B& not A1&A))
                                                                                    ^^^
SyntaxError: invalid syntax

infinitymdm, 2023-12-15:
I think this happens because python operator precedence differs from verilog operator precedence, and the current parsing may not isolate negated terms correctly. Looking into it.

thesourcerer8, 2024-04-20:
Are you going to fix CharLib to support Verilog syntax, or is the Python syntax the preferred Syntax, and I should generate Python syntax?

infinitymdm, 2024-04-20:
This is something that definitely should be fixed, I just haven't had the time to get around to it. For the time being, if you want to use Charlib you'll have to use the Python syntax.

thesourcerer8, 2024-05-03:
Ok, we fixed the Python syntax from our side, so from my side we could close this issue, but we could also leave it open if you still want to implement Verilog syntax.

infinitymdm, 2024-05-03:
> Ok, we fixed the Python syntax from our side, so from my side we could close this issue, but we could also leave it open if you still want to implement Verilog syntax.

I still consider this a bug in CharLib that needs to be fixed. It should work according to the usual standards for order of operations. Glad you were able to find a workaround for this though.

PR #62 (2025-06-27): Logic Parsing Improvements
This PR includes several improvements to CharLib's logic function parsing:
- Operator precedence is now handled as specified in IEEE 1800-2023
- Logical negation is now interpreted correctly (fixes #23)
- The XNOR operator (~^) is now supported
    - Note that the alternative XNOR operator "^~" is not supported, as this increases complexity for parsing negations
closed 2025-06-27 by infinitymdm