stineje/CharLib #33: dependency?

thesourcerer8, 2024-04-24:
When trying to install a newer lclayout version, I get the following error from pip:
charlib 1.0.0 requires liberty-parser<0.0.17,>=0.0.16, but you have liberty-parser 0.0.23 which is incompatible.
I can only see the ^0.0.16 requirement in the GitHub repo. Do you have any idea where the <0.0.17 requirement comes from and why?

infinitymdm, 2024-04-24:
This does turn out to be an issue with the poetry configuration. The dependency for liberty-parser is specified as `^0.0.16`, which would allow updates to the next version... except that the major and minor version are both zeros, so there's some odd behavior. See https://python-poetry.org/docs/dependency-specification/ for more info. The TLDR is that the current release says it _requires_ 0.0.16, even though it works with >=0.0.16. 

Future releases will correct this to have less strict version requirements. Hopefully we can get a new release out in the next few days, and this will be fixed.

thesourcerer8, 2024-05-03:
I tried installing it on Jupyter and Google Colab, and got dependency issues, e.g.:
[image]
I definitely need full interoperability between CharLib and lclayout and lctime, and venv's aren't a solution for Jupyter/Colab and perhaps even other situations.

PR #40 (2024-05-06): Bump patch version and fix liberty-parser dependency
Fixes #33 

- Bump patch to .1
- Fix the dependency on liberty-parser to allow v0.0.x rather than exclusively 0.0.16
closed 2024-05-06 by infinitymdm