stineje/CharLib #37: Allow choosing cells from CLI

infinitymdm, 2024-04-24:
Allow users to choose which cells to run characterization against from the command line. This could be implemented as a regex filter, so that users can give full cell names to select specific cells or regex to match larger groups of cells.

thesourcerer8, 2024-04-29:
From my point of view, this could be solved by simply providing .yml files with the specific selection of cells that should be characterized. I don't think that a further selection is really necessary, but it also wouldn't hurt to have it.

thesourcerer8, 2024-04-29:
Perfectly fine for me to support both use-cases then.

stineje, 2024-04-29:
Let us know any more great enhancements and suggestions.  Your comments are extremely super!  Marcus is awesome and we have big plans on expanding.  Take care.

All my best,

James

closed 2024-05-07 by infinitymdm