stineje/CharLib #38: Prevent concurrent runs from interfering with one another

infinitymdm, 2024-04-24:
Concurrent characterization runs using the same config file currently write out results to the same filename, overwriting previous results. This means that the results of all but the final characterization run are entirely lost. Provide a mechanism to allow the results of concurrent runs to be written to different paths.

(There may also be an issue here re. ngspice, as the ngspice-shared PySpice backend uses a singleton for the ngspice instance. This may take some looking into, or it might be a non-issue.)

thesourcerer8, 2024-04-29:
From my point of view, we need the possiblity to just specify a single .yml file (which contains a single cell) on the commandline. I currently dont see a need to run several characterizations concurrently for the same cell.

infinitymdm, 2024-08-21:
This should have been closed with #41; not sure why it didn't automatically close. You can specify where you want the output liberty file with the `-o` switch, and cells can be selected either by a) specifying a path to a config file with just the cells you want or b) using the `-f` switch to filter a larger yaml file for the cells you want to run.

closed 2024-08-21 by infinitymdm