stineje/CharLib #42: Add Usage guide

infinitymdm, 2024-05-07:
With the new CLI features recently added, there are quite a few undocumented features of CharLib. Add a guide showing how to use CharLib's CLI features.

Blebowski, 2025-06-03:
Hi,
I am trying to bring-up the `Charlib` on a proprietary standard cell library.

I have couple of observations which may be fitting here:
1. The default guide for installation shows that you should install the "PySpice" from a GIT URL. I did follow these steps, but my code did not work with following error:
```
charlib -h
Traceback (most recent call last):
  File "/tmp/CharLib/venv/bin/charlib", line 5, in <module>
    from charlib.characterizer.run import main
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/charlib/characterizer/run.py", line 12, in <module>
    from charlib.characterizer.Characterizer import Characterizer
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/charlib/characterizer/Characterizer.py", line 8, in <module>
    from charlib.characterizer.TestManager import TestManager, CombinationalTestManager, SequentialTestManager
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/charlib/characterizer/TestManager.py", line 11, in <module>
    from PySpice.Tools.StringTools import str_spice
ModuleNotFoundError: No module named 'PySpice.Tools.StringTools'
```
I tried to search a little, and it seems that the packaged version of `charlib` available on my platform did rely on some older version of PySPICE, while the GIT repo gave me the latest. In general, I think it is bad idea to combine "packaged" Python dependency pushed into sth like `pip` with HEAD of a custom repository fork. It is likely to break often. Either, everything should be a packaged dependency, and downloaded via pip, or a user can package the solution by himself from the latest master of `charlib`. What is the reason behind having the forked version of PySPICE so much "ahead" than the original repository.
2. Related to the first topic, It would be good to add option to package the application by myself. Just a few commands to launch poetry in a new virtual env, run `poetry install` and then the `pip` install for the forked PySPICE. That-way bring-up will be simpler.
3. I do have NGSPICE 44.2 installed and availalble in PATH, which according to the PySPICE commits should be enough. But apparently, according to the error message I get:
```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib64/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/charlib/characterizer/Characterizer.py", line 54, in characterize_cell
    raise e
  File "/tmp/CharLib/charlib/characterizer/Characterizer.py", line 49, in characterize_cell
    return cell.characterize(self.settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/charlib/characterizer/TestManager.py", line 311, in characterize
    input_capacitance = self._run_input_capacitance(settings, pin.name) @ u_F
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/charlib/characterizer/TestManager.py", line 281, in _run_input_capacitance
    simulator = Simulator.factory(simulator=settings.simulator)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/PySpice/Spice/Simulator.py", line 121, in factory
    obj = sub_cls(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/PySpice/Spice/NgSpice/Simulator.py", line 92, in __init__
    self._ngspice_shared = NgSpiceShared.new_instance()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/PySpice/Spice/NgSpice/Shared.py", line 457, in new_instance
    instance = cls(ngspice_id=ngspice_id, send_data=send_data, verbose=verbose)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/PySpice/Spice/NgSpice/Shared.py", line 486, in __init__
    self._init_ngspice(send_data)
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/PySpice/Spice/NgSpice/Shared.py", line 569, in _init_ngspice
    rc = self._ngspice_shared.ngSpice_Init(self._send_char_c,
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/cffi/api.py", line 914, in __getattr__
    make_accessor(name)
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/cffi/api.py", line 910, in make_accessor
    accessors[name](name)
  File "/tmp/CharLib/venv/lib64/python3.11/site-packages/cffi/api.py", line 840, in accessor_function
    value = backendlib.load_function(BType, name)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function/symbol 'ngSpice_Init' not found in library '<None>': python3: undefined symbol: ngSpice_Init
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/projects/Digital_flow/work/oille/CharLib/./charlib/characterizer/run.py", line 296, in <module>
    main()
  File "/projects/Digital_flow/work/oille/CharLib/./charlib/characterizer/run.py", line 66, in main
    args.func(args)
  File "/projects/Digital_flow/work/oille/CharLib/./charlib/characterizer/run.py", line 157, in run_charlib
    library = characterizer.characterize()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
It seems that I should have the ngspice built as shared object lib. But this is not apparent from any "bring-up" / "how-to" step-by-step guide on how to use `charlib`. I mean sure, one can dive into documentation of PySPICE, but that is something one should not do if he simply wants to use `charlib`. That should be hidden from user, since PySPICE is simply a dependency of `charlib`. APIs of `charlib` are: CLI commands and YAML syntax. Everything else should be hidden by a good "how to bring-up" guide. Otherwise people will easily get lost and not pursue with using the tool.

I can contribute some of the documentation changes once I get the `charlib` up and running.

Blebowski, 2025-06-05:
Hi @stineje,

could you please help me with the bring-up ?

My setup is as following:
1. I cloned the repo - I would like to take the Github version, not some packaged version in PIP.
2. I created a local Python virtual env and launched it.
3. I installed poetry into the virtual env.
4. I installed all poetry dependencies. (`poetry instal`) into the virtual env.
5. I am trying to launch `./run.py run` but I still get:

```
AttributeError: function/symbol 'ngSpice_Init' not found in library '<None>': python3: undefined symbol: ngSpice_Init
```

Now I have NGSPICE built as shared library. Install location of my ngspice is:
```
/tools/opensource/ngspice/44.2
```

I went through documentation of PySPICE:
```
https://pyspice.fabrice-salvaire.fr/releases/v1.5/faq.html
```

and found out, that I probably need to tell PySPICE where the ngspice executable and shared library are located.

So, in `./run.py` I have imported the underlying PySPICE module:
```
import PySpice.Spice.NgSpice.Shared
```
and then tried to set the path in the `main()`:
```
PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH = "/tools/opensource/ngspice/44.2/lib/libngspice.so"
PySpice.Spice.NgSpice.Shared.NgSpiceShared.NGSPICE_PATH = "/tools/opensource/ngspice/44.2/bi/ngspice"
```
I still get the same error as above.

I think `Charlib` should have a command line switch that points it to the ngspice executable or binary to be used,
so that one does not need to "hack it" just to get it working.

infinitymdm, 2025-06-05:
@Blebowski what OS/distro are you using?

Usually the way to install from the latest git version is to run `pip install git+https://github.com/stineje/CharLib.git` rather than using poetry. The end result should be the same though, which makes me think this is an issue with your ngspice install. Or at least that PySpice doesn't like something about it; I've never seen this particular issue before, so I'm not sure. You've provided good instructions to replicate the issue, so I'll see if I can do so.

I definitely agree with you that the packaged/unpackaged stuff needs some work; the only reason we ended up doing it this way is that PyPI doesn't allow us to use our customized version of PySpice directly as a dependency in the pip packaging. They have good reasons for that, but those aren't particularly relevant here; what matters is we had to find some alternative solution.

As far as why we need a customized version of PySpice, we use .measure directives for lots of simulation results. PySpice didn't have support for passing .measure results back to the user, so our customized version fixes that as well as tracking the -devel branch of upstream PySpice. There are a bunch of other small fixes we've made, all of which have been submitted to upstream PySpice, but none of which have been accepted.

I also agree with you about the ngspice install being out of scope for CharLib. That's exactly why we don't provide install instructions: the normal installation methods provided by your distro package or ngspice directly should be sufficient, and PySpice should handle finding the executable code at runtime.

Blebowski, 2025-06-05:
I managed to move further.
After diving into the code-base of PySPICE, it turns out it tries to locate the NGSPICE shared library by means of:
`ctypes.util.find_library`. Which may behave differently on various platforms (according to StackOverflow).
PySPICE documentation says something like:
```
You can set globally the default simulator using the attribute 
```
If the default setting doesn’t match your environment, then you have to fix globally the attribute [PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH](https://pyspice.fabrice-salvaire.fr/releases/v1.5/api/PySpice/Spice/NgSpice/Shared.html#PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH). Note you have to place a brace pair just before the extension, for example C:\...\ngspice{}.dll.

You can also fix the value of [PySpice.Spice.NgSpice.Shared.NgSpiceShared.NGSPICE_PATH](https://pyspice.fabrice-salvaire.fr/releases/v1.5/api/PySpice/Spice/NgSpice/Shared.html#PySpice.Spice.NgSpice.Shared.NgSpiceShared.NGSPICE_PATH).
```

But that does not work (at least for me). Then the PySpice looks at the `NGSPICE_LIBRARY_PATH` env var, and does
some prefix stripping before passing it to the `ctypes.find_library`. After looking at examples of calls to `find_library`
on Stack-Overflow, following combination worked:
```
export LD_LIBRARY_PATH = /tools/opensource/ngspice/44.2/lib
export NGSPICE_LIBRARY_PATH = ngspice
```
Adding the `.so` , or adding the `{}` as the manual suggested did not help.

TBH, it would be much simpler if there was just a simple way on how to explicitly override path to the library. Such switch would be propagated to the `charlib` and could be set by a CLI option.

infinitymdm, 2025-07-04:
Fixed by #65

closed 2025-07-04 by infinitymdm