stineje/CharLib #47: Deprecated PySpice API

TiMo3654, 2025-04-04:
As one can see in the terminal output below, the file setup and read-in seems to work. However once the simulation should start, the NameError below occurs.

This is again a difference between [adapted](https://github.com/infinitymdm/PySpice/blob/master/PySpice/Spice/Netlist.py) and [original](https://github.com/PySpice-org/PySpice/blob/master/PySpice/Spice/Netlist.py) PySpice versions. 

My setup can be found in https://github.com/TiMo3654/skywater-std-cell-char/tree/main.

Any help on this would be greatly appreciated!

(charlib) ed-srv-plasma $ charlib run ./
Searching for YAML files in ./
Reading configuration found in "sky130_fd_sc_lp__inv_1.yml"
Recognized SKY130_FD_SC_LP__INV_1 pin Y function as INV
Running input_capacitance for pin A of cell SKY130_FD_SC_LP__INV_1
Traceback (most recent call last):
  File "/home/moldenhauert/miniconda3/envs/charlib/bin/charlib", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/run.py", line 60, in main
    args.func(args)
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/run.py", line 127, in run_charlib
    library = characterizer.characterize()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/Characterizer.py", line 39, in characterize
    cells = [self.characterize_cell(cell) for cell in self.tests]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/Characterizer.py", line 48, in characterize_cell
    return cell.characterize(self.settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/TestManager.py", line 307, in characterize
    input_capacitance = self._run_input_capacitance(settings, pin.name) @ u_F
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/TestManager.py", line 280, in _run_input_capacitance
    simulator = circuit.simulator(temperature=settings.temperature,
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/Netlist.py", line 745, in simulator
    raise NameError("Deprecated API")
NameError: Deprecated API

infinitymdm, 2025-04-04:
Thanks for bringing this to our attention! We're aware of recent PySpice changes, and we're working on updates to CharLib. As it so happens, we've been preparing a sky130 test configuration as well (see #45).

Could you try running with the latest git version of CharLib? You can install it in your venv with `pip install git+https://github.com/stineje/CharLib`. I *think* this should be fixed there, we just haven't drafted a new release yet.

infinitymdm, 2025-04-07:
Could you share the full traceback? You might also try running with the `--debug` flag. That will print a bit of extra information and write generated spice files to ./debug. 

I'm also going to attempt to replicate your setup and see what I can figure out.

infinitymdm, 2025-04-07:
Again, thanks for posting the issue. This sort of feedback is how we improve CharLib.

My takeaways so far:
- It would probably be nice to have some built-in tests to make sure the `models` key is set correctly, so that issues like this can be more easily identified (those Python tracebacks typically aren't super helpful)
- We need to do something different re. PySpice/CharLib alignment
- Test-circuit-level multithreading would make CharLib a lot more usable

Anything you think I'm missing?

TiMo3654, 2025-04-08:
Thank you for your active support! I am really interested in this project because I want to have an easy to modify, share and execute tool pipeline to do some data exploration on standard cell characterization data.

Unfortunately, your fix proposal has not worked so far. I have pushed your proposed changes in my repository and on executing charlib I get the following error trace:

(charlib) ed-srv-plasma $ charlib run ./
Searching for YAML files in ./
Reading configuration found in "sky130_fd_sc_lp__inv_1.yml"
ngspice-shared
Running input_capacitance for pin A of cell SKY130_FD_SC_LP__INV_1
Warning: Optran step size potentially too large.
Warning: m=xx on .subckt line will override multiplier m hierarchy!
Using KLU as Direct Linear Solver
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/Characterizer.py", line 49, in characterize_cell
    return cell.characterize(self.settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/TestManager.py", line 311, in characterize
    input_capacitance = self._run_input_capacitance(settings, pin.name) @ u_F
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/TestManager.py", line 297, in _run_input_capacitance
    analysis = simulation.ac('dec', 100, f_start, f_stop)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/Simulation.py", line 768, in wrapper
    return self._run(analysis_method, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/Simulation.py", line 758, in _run
    _ = self._simulator.run(self)
        ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/NgSpice/Simulator.py", line 118, in run
    measurements = self._ngspice_shared.run()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/NgSpice/Shared.py", line 1187, in run
    results = self.exec_command(command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/PySpice/Spice/NgSpice/Shared.py", line 854, in exec_command
    raise NgSpiceCommandError(f"Command '{command}' failed")
PySpice.Spice.NgSpice.Shared.NgSpiceCommandError: Command 'run' failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/Characterizer.py", line 51, in characterize_cell
    if self.omit_on_failure:
       ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Characterizer' object has no attribute 'omit_on_failure'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/moldenhauert/miniconda3/envs/charlib/bin/charlib", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/run.py", line 66, in main
    args.func(args)
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/run.py", line 157, in run_charlib
    library = characterizer.characterize()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/site-packages/charlib/characterizer/Characterizer.py", line 37, in characterize
    cells = pool.map(self.characterize_cell, [*self.tests])
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moldenhauert/miniconda3/envs/charlib/lib/python3.12/multiprocessing/pool.py", line 774, in get
    raise self._value
AttributeError: 'Characterizer' object has no attribute 'omit_on_failure'
(charlib) ed-srv-plasma $  

The fourth line from the top is a print statement which I have added in Characterizer.py to try out different ngspice options (ngspice-shared, ngspice-subprocess) in the config yaml.                       

I suppose that the first exception comes from the PySpice/ngspice interactions problems which I have also without charlib.

The second exception originated from charlib even though I do not understand the cause, because I do think that in line 68 in Characterizer.py this attribute is initialized in the constructor.

TiMo3654, 2025-04-08:
Might be linked to https://github.com/stineje/CharLib/pull/44

infinitymdm, 2025-04-08:
Indeed, looks a bug was introduced in #44. Characterizer.py L51 should have `self.settings.omit_on_failure`. Note the missing `.settings`, which accesses the `CharacterizationSettings` object. @deserthobo13 do you mind throwing together a quick fix for that?

In your case though, something else is causing the error, since L51 only gets run when there's an exception during the characterization. Not sure what's causing that, nor why this runs on my machine but not on yours.

Just in case we're running into a version-specific bug, here's what I'm running:
- ngspice: 44.2 (Solus package)
- Python: 3.11.11 (Solus package)
- PySpice: 1.6 (from https://github.com/infinitymdm/PySpice)

I'm hoping this isn't a packaging issue with ngspice or Python, but it's possible. Let me know if you have any guesses.

TiMo3654, 2025-04-09:
@infinitymdm I was finally able to run CharLib successfully in my ubuntu container with the ngspice and addons from the package manager, your PySpice Version and the most recent CharLib version installed from github.

My problems were therefore OS related, but nothing of that originated from CharLib. The problem comes from the ngspice shared library and PySpice interaction. Maybe sometime I will have the time to look into that, but for now I am happy with my container.

Thanks for all your assitance! I hope, that I can contribute to the project.