stineje/CharLib #60: SRAM characterization support

Blebowski, 2025-06-13:
**Is your feature request related to a problem? Please describe.**

Digital designs often contain hard-macro SRAMs. 
SRAM memories are usually synchronous with the clock domain they are used in.
Thus correctly characterized timing is needed for such memories.

The memories behave similarly as a flop in the regard that:
1. A setup and hold times for input ports needs to be measured.
2. A CK -> DATA Out delay needs to be measured.

The complexity of memory characterization is much higher because:
1. When measuring setups / holds, each input bit of an input pin must be measured separately.
2. When measuring output delay, a maximal value must be measured (potentially, delay from each memory location must be measured per each pin of output data.
3. Likely a whole memory must be simulated at once -> Long simulation time -> Thousands of transistors.

However, typically only one memory is characterized at single time, and a single liberty
file exists for a single memory. So, paralellization could be heavily exploited across multiple
cores.

stineje, 2025-06-13:
Hi,

SRAM characterization is a more difficult problem, because you don’t have the FUNCTION within the Liberty file for a SRAM.  This means, you cannot just say A*B or 1*1 will cause an output of 1.  It’s highly dependent on any input.

This ultimately means, figuring out 1.) what vectors to use 2.) what is in the memory as stored values complicates the timing. I do plan on looking at this sometime depending on funding levels.  We are currently working with Synopsys using PrimeLib to use commercial tool usage - that’s, of course, a commercial tool - I know Siemens has a tool too.  I plan on doing the same for CharLib.

I think this is a great future idea.  Take care.

All my best,

James

Blebowski, 2025-06-13:
Yeah, I think this is rather ultimate goal, so really take it as a wish-list please.
I think there is plenty of things to do in this project before tackling such big thing.
Anyway thanks for reply.