stineje/CharLib #61: Flip flop characterization problems.

Blebowski, 2025-06-13:
**Problem Description**
I am trying to characterize a simple flip-flop from a commercial PDK.
It is not a scan flip flop. It has no set / reset asynchronous pins.
I see two problems:
1. The same set of "slews" is used for delay as for "setup/hold" measurement. In the technolgies I work with, the delay matrix is 7x7, but setup / hold timing has only 3x3. It would be good to split the matrix for IO delays (`input slew` , `output cap`) from the setup / hold matrix (`constrained pin slew` , `related pin slew`). Now the `slew` is taken for both which is confusing. E.g. from the liberty file:
```
 /* Table Templates */
  lu_table_template (delay_template_3x3) {
    variable_1 : input_net_transition;
    variable_2 : total_output_net_capacitance;
    index_1 ("0.002, 0.162966, 0.8");
    index_2 ("0.001, 0.00540346, 0.0228305");
  }
  lu_table_template (setup_template_3x3) {
    variable_1 : related_pin_transition;
    variable_2 : constrained_pin_transition;
    index_1 ("0.002, 0.162966, 0.8");
    index_2 ("0.001, 0.00540346, 0.0228305");
  }
```
2. When I try to characterize a flip-flop, I get vastly different results compared to FAB provided data
at the same operating conditions (TT, 25 C, 1.2V):

E.g. charlib result for D pin setup:
```
rise_constraint (setup_template_3x3) {
          index_1 ("0.002, 0.162966, 0.8");
          index_2 ("0.001, 0.00540346, 0.0228305");
          values ( \
            "1.402000, 1.402000, 1.402000", \
            "1.402000, 1.402000, 1.402000", \
            "1.402000, 1.402000, 1.402000");
        }
```
While if I look at the setup time of D pin for (0.002, 0.002) from foundry provided `.lib`
file, I get sth like 0.046. The difference is something like factor of 35.

Further, all the setup times are the same, which is not the same in the foundry provided
data.

I do not have SPICE netlist extracted with parasitics from layout of the cell.
I only have "LVS" netlist -> List of transistors and their connections.
Maybe since the measurement occurs at range where circuit is metastable,
the output delay may significantly vary. But I don't know.

Do you have any clue on what may have went wrong ?

**Expected behavior**
Be within some reasonable range from the FAB characterized `.lib` file (e.g. +-20%)

stineje, 2025-06-13:
This error is likely due to the search space.  Normally, characterization tool use a binary search to find these points.  If the space is not given correctly.  Check your matrix to make sure it’s correct.  There are several ways to validate this but I tend to like to check input capacitance of an inverter to see things like this.   I would be interested if you change the matrix that things will behave differently.  Thanks for the question and the post!

  1.  When I try to characterize a flip-flop, I get vastly different results compared to FAB provided data

infinitymdm, 2025-06-16:
I'm pretty confident this is a bug in CharLib. Currently the setup/hold search is accomplished via a pair of binary searches that ping-pongs back and forth between the two constraints, decreasing them until things stop working. The algorithm is basically:

0. Givens: range of setup times to search, range of hold times to search, precision threshold
1. Set hold time to max in range.
2. Bisect setup time range & test. Does the cell still change state as expected? If so, replace setup search range with the half below the bisection. If not. replace setup search range with the half above the bisection.
3. Set setup time to max in range.
4. Bisect hold time range & test. Does the cell still change state as expected? If so, replace hold search range with the half below the bisection. If not. replace hold search range with the half above the bisection.
5. Repeat from 1 until both ranges are smaller than the precision threshold.

This process is repeated for each combination of slew and load given, but as identified in #28 we don't currently handle clock slews correctly. You can take a look at the code for this [here](https://github.com/stineje/CharLib/blob/a4772e4b23c2aca253af60355f310d23430b31bd/charlib/characterizer/TestManager.py#L806). Other than the clock skews, does anything look particularly incorrect there?

@Blebowski thanks for your feedback here. It's a huge help for us in identifying these sorts of issues.

infinitymdm, 2025-06-17:
Sorry, looks like my info was out of date. The algorithm I posted above is incorrect; I think that was what we were doing before I did some additional reading.

Here's the paper that I based the current code on (also linked in the code): https://ieeexplore.ieee.org/document/4167994. Fig. 7 has a really nice plot of the CLK->Q delay surface, where the metastability boundary is shown as a semi-asymptotic set of peaks (exactly what you're talking about re. potentially infinite delay). The authors suggest choosing a 10% degraded CLK->Q contour, then applying a piecewise linear interpolation. The left half of the flowchart in Fig 6. is the process we need.

Clearly there's something wrong with the code as-is; for one thing, the PWL interpolation should be giving us different values for each setup/hold pair. That's not happening for some reason; the result table you posted before has the same value in every slot. We probably have an incorrect implementation of that algorithm, since I still see some binary-search stuff [in the code here](https://github.com/stineje/CharLib/blob/a4772e4b23c2aca253af60355f310d23430b31bd/charlib/characterizer/TestManager.py#L806), in `_sweep_ts` and `_sweep_th`.

infinitymdm, 2025-07-25:
Discussed this with @stineje. We're going to rewrite the sequential characterization procedures to use the less complex but better-proven binary search algorithm I mentioned above on Jun 16. This still won't be amazingly accurate; other open characterizers report error of around 400% with this method vs. commercial tools. But 4x is quite a bit better than 30-50x.

Long-term, I'd like to make these methods more modular, so that users can select which algorithm they want to use for a particular procedure, or even roll their own procedures so long as the interface matches up. That should also open the door to using CharLib as a platform for characterization experiments, where authors can bring their own simulation methods, test them, and easily compare with other methods. We've got a long way to go before we're at that stage, but that's part of my vision here.