stineje/CharLib #66: [Feature Request] User messages - Use logger instead of print

Blebowski, 2025-07-02:
**Is your feature request related to a problem? Please describe.**
The current format of user messages is quiet confusing.

If some (e.g. error message) is printed, usually Charlib prints also the Python stack-trace.
Although the actual message is just at the end, it seems confusing and prevents quickly
finding "why the error occurs". I think any situation in Charlib should resolve in one of:
- Charlib executes, prints info messages, normal operation
- Upon error, Charlib throws error and quits.  There is no stack-trace.
Showing stack-trace is generally considered a bug in the SW.

Some errors could be "supressed", or there could be a way of "softly ignoring" some error
(e.g. currently there is the "skip cells upon error" functionality).

**Describe the solution you'd like**

All this stuff is already resolved by Python `logging` library. IMHO, we would be enough
with 5 basic logging levels:
- DEBUG - By default disabled, enabled by "Debug"
- INFO - By default enabled. Shows basic progress.
- WARNING - Untypical situation occured, proceed normally.
- ERROR - A problem occured. Count ocurrence of problems if more than e.g. X problems occur -> Quit. Have option to configure X. X=0 would mean: any error means quit.
- FATAL - Can't continue, quit immediately.

IMHO the logging could somewhat map to the underlying PySpice logger, e.g. like so:
- DEBUG - Pyspice level is INFO
- INFO - Pyspice level is  WARNING
- WARNING - Pyspice level is  ERROR
- ERROR - Pyspice level is  FATAL
- FATAL - Pyspice level is FATAL

In other words, Pyspice level would be +1 compared to the default one.

infinitymdm, 2025-07-04:
This would be a big improvement to clarity when something goes wrong in CharLib, and would enable us to add some extra levels of `-v` and `-q` flags as well. 

I like the idea of the PySpice logging level being one level below CharLib, but the one-to-two mapping of PySpice FATAL to CharLib ERROR and FATAL is probably going to be tricky. 

There are some cases where multiple test vectors (i.e. with different masking conditions) cover the same test arc (e.g. for an OR gate, input pin A to output pin Y, positive unate is a valid arc regardless of the state of pin B), so for those we can have PySpice FATALs propagate to ERROR and try alternative test arcs. Likewise, when we have `omit_on_failure: true`, PySpice FATALs should probably map to ERROR.

Actually, since our simulations are always on a per-cell basis, it probably makes sense to map _all_ PySpice FATALs to ERRORs. The only exception should be cases where PySpice can't run at all for some reason, such as when it can't find a simulator.

infinitymdm, 2025-07-04:
Related to #31