stineje/CharLib #74: [Feature Request] Infer I/O pins from cell functions

infinitymdm, 2025-07-18:
**Is your feature request related to a problem? Please describe.**
Specifying functions on cells is a tedious enough task on its own, and provides all the information CharLib should need to identify all input and output pins (aside from power & wells, which are defined with `named_nodes` anyways). 

Instead of requiring users to specify `inputs` and `outputs` for each cell, infer them from `functions`. The information is already there - we just have to parse it out. The `Function` class already has this feature in the `operands` accessor - we just have to make use of it.

**Describe the solution you'd like**
Make `inputs` and `outputs` on `cell` configs optional, and instead infer the inputs from `functions`.

Blebowski, 2025-07-19:
Does this really have added value for user ? If someone is designing a PDK, they likely have this information
somewhere in some format, so it is just a matter of propagating it to the YAML file.

IMHO, it is similar as the other issue with the automated function recognition.

Rather there could be some forms of "consistency checks". E.g. warnings / errors on:
- If a pin is specified in YAML file, but it does not exist in the SPICE model
- If a pin exists in SPICE model, but it is not specified in the YAML file
- If a pin occurs within the "function", but it is not defined within inputs or outputs (and vice-versa).

infinitymdm, 2025-07-24:
@Blebowski This issue and #56 are very closely related, but offer different solutions to slightly different problems.
- If functions are extracted automatically, CharLib still needs to know which pins are `inputs` and which are `outputs`. #56 would allow leaving out the `functions` key from a cell configuration, which is useful if you don't know what a cell does for some reason, or if the documentation is wrong.
- If functions are provided, the `inputs` and `outputs` keys are redundant; the information is already there in the functions. This issue (#74) seeks to allow leaving out those two keys instead.

Definitely agree with you on the need for consistency checks. I am of the opinion that several of CharLib's data structures need a rework, to add these and improve modularity. That, however, is a discussion for another issue.

closed 2025-12-17 by infinitymdm