stineje/CharLib #81: Documentation should clarify how to install ngspice-shared

afreynxp, 2025-08-06:
**Problem Description**
Default simulation engine is ngspice-shared so CharLib won't work if  only `ngspice` is installed.
The error message is obscure (missing `ngspice_Init` method).

**Expected behavior**
The documentation should explain how to install the simulation engine.
For example, for ngspice-shared, on Ubuntu, you have to do:
```
sudo apt install libngspice0-dev
```

Additionally, the error message could indicate more clearly that the engine is not available.

infinitymdm, 2025-08-12:
It would certainly improve things to point out that the current builds pretty much require ngspice-shared. Documentation PRs are always welcom

IMO the poor error messages are an upstream issue, as they propagate up from PySpice. This should probably be posted on the PySpice repository instead: https://github.com/PySpice-org/PySpice. From a usability standpoint I definitely see where you're coming from, but PySpice doesn't give downstream developers (like me!) enough context to actually tell what's going on when it propagates this error. To make matters worse, the error message you're mentioning is one I've only seen when ngspice actually *is* built with the shared library, but the library is in a non-standard location. 

If by chance you're seeing this error on a Chipathon 2025 docker image, the solution can be found here: #71

Blebowski, 2025-09-28:
Hi @infinitymdm ,

I think there could be small logic that tries to figure out if the `ngspice-shared` is available.
I know from developers perspective, it is an "upstream problem", but from users perspective, it
is a problem that causes bad user experience for Charlibs users.

infinitymdm, 2025-10-06:
For the time being I would prefer a solution that lives solely in the documentation rather than in the code, especially since we intend to move away from PySpice long-term. That said, I don't disagree that a runtime check for the shared library would be an improvement.

PRs are welcome, as always.

Blebowski, 2025-10-07:
Huge +1 for:
```
especially since we intend to move away from PySpice long-term. 
```